ros-tooling/aws-roadmap #15: [2d] ROSBag2 Compression Investigation

emersonknapp, 2019-10-08:
## Description

rosbags generated by `ros2 bag record` are much larger than corresponding bags in rosbag. As per the Apps team, this makes rosbag2 unusable in RoboMaker right now because of high cost of bag storage that will lead to bad customer experience. The goal of this spike is to understand if compression can be used in rosbag2 and if/how much it will help RoboMaker rosbag2 support.

* Need to get what success criteria looks like in terms of bagfile size
* Need to understand how bagfiles are generated in rosbag v. rosbag2

## Spike Timeboxing

2 days

## Spike Outcome
 * Stories: no
 * 1-pager: yes, detail discussions with Simulation team: https://quip-amazon.com/05dHALqcV9b2/Spike-rosbag2-compression

## Goals

* [ ] This spike has been updated with the indings

dabonnie, 2019-10-08:
FYI @thomas-moulard  How should we handle company internal links? Usually when we share quip docs it is internal only, so link sharing when public facing may not be desirable.

emersonknapp, 2019-10-08:
From Devin

> RoboMaker records everything into a rosbag and dumps output to an S3 bucket
> 
> - this takes too much disk space in ros2
> - was compression used in ROS1?
> - this is asking for rosbag2 --compress to compress on the fly

thomas-moulard, 2019-10-08:
@dabonnie I added a wiki -> https://github.com/ros-security/aws-roadmap/wiki/%5B2d%5D-ROSBag2-Compression-Investigation-%2315

dabonnie, 2019-10-09:
Updated with this [merge](https://github.com/ros-security/aws-roadmap/commit/8d0c9a9a0eeaee0d98a3a326ddc5443a2df187b0)

closed 2019-10-09 by dabonnie