ros-tooling/aws-roadmap #27: Measure and average system CPU %

emersonknapp, 2019-10-08:
## Description

Measure linux system CPU % using /proc/stat (example found [here](https://github.com/aws-robotics/health-metrics-collector-ros1/blob/master/health_metric_collector/src/cpu_metric_collector.cpp)). Store the resulting measurement in a [RunningStatistics](https://code.amazon.com/packages/B9Ros2TopicStatistics/blobs/mainline/--/statistics/include/statistics/moving_average/running_statistics.hpp) averaging container. 

Note: the RunningStatistics class will need to be migrated to the utilties project defined in [ARF-3843](https://issues.amazon.com/issues/ARF-3843).

## Test Plan

Unit tests for new classes and fake data can be used to simulate /proc/stat CPU information
## Documentation Plan

Document code

## Release Plan

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [Design Doc](https://quip-amazon.com/KzzfAUQeRkVY/WIP-Design-Doc-ROS2-Runtime-Metrics-System-Metrics)

dabonnie, 2019-11-05:
Opened PR for the collector: https://github.com/ros-tooling/system_metrics_collector/pull/4

dabonnie, 2019-11-08:
Opened PR for Periodic API: https://github.com/ros-tooling/system_metrics_collector/pull/6

dabonnie, 2019-11-12:
Opened PR which adds measurement mechanism: https://github.com/ros-tooling/system_metrics_collector/pull/11

closed 2019-11-20 by dabonnie