ros-tooling/aws-roadmap #60: As a ROS2 developer, I want log files of all ROS2 logging, so that I can easily inspect logs offline

emersonknapp, 2019-10-10:
## Description

In the Dashing release, there ended up being problems with logging going out to files due to issues with `log4cxx`. 

To fix this, work has been taken on to replace `log4cxx` with a different logging backend `spdlog`. This feature has now been merged, but we need to make sure the feature is usable and documented to our satisfaction.

* https://github.com/gabime/spdlog
* https://github.com/ros2/rcl_logging/pull/17

## Documents

TODO

## Completion Criteria

* [ ] ROS2 users can have their nodes' logging output to files
* [ ] It is documented how to configure/use this functionality

thomas-moulard, 2019-10-11:
I believe this has already been completely done. We should take a look at what's getting done and see if there is actually any work left.

piraka9011, 2019-11-21:
Last I checked on my Mac, `~/.ros/log` has a bunch of `.log` files that log the output of running ros2 commands like launch.

Docs are [here](https://index.ros.org/doc/ros2/Tutorials/Logging-and-logger-configuration/) and [here](https://index.ros.org/doc/ros2/Concepts/Logging/)

thomas-moulard, 2020-01-14:
No implementation is required for this as this is most likely already meeting our bar. If we don't get any particular request by end-of-week (01/17), my recommendation is to define a doc to test the API (i.e. log data using different calls, ensure that enabling `spdlog` works fine) and make sure everything works as intended on `master`.

piraka9011, 2020-02-03:
PR [here](https://github.com/ros-security/aws-roadmap/pull/175)

thomas-moulard, 2020-02-14:
Moved back to in progress until we review this.

closed 2020-02-14 by thomas-moulard