ros-tooling/aws-roadmap #62: As a ROSbag2 user, I need rosbag2 recording to meet a specific performance benchmark, so that I have plenty of system resources for my business logic

emersonknapp, 2019-10-10:
## Description

"As a ROSbag2 user, I need rosbag2 recording to meet a specific performance benchmark, so that I have plenty of system resources for my business logic"

rosbag2 performance has been reported as not being good enough currently. Bring its performance for the use case into line with required benchmarks.

## Documents

TODO - document
1. Use case
1. Current measured performance
1. Desired performance

## Completion Criteria

* Negotiated benchmark goal has been completed

emersonknapp, 2019-10-10:
Iceboxing until more concrete use case and goals are negotiated

zmichaels11, 2019-12-13:
@piraka9011 write user stories for this epic

piraka9011, 2019-12-13:
Metrics we want to track:

* Disk usage
* Memory usage
* CPU usage
* Database transaction time (R/W)
* Compression time
* Playback time

dabonnie, 2020-02-06:
Closing this in favor of https://github.com/ros-tooling/aws-roadmap/issues/150. We currently have metrics defined to measure performance and an internal review meeting to discuss said metrics today.

closed 2020-02-06 by dabonnie