ros-tooling/aws-roadmap #63: As a ROS2 user, I want to never encounter a deadlock when using ROS2 services, so that I can build reliable ROS2 applications

emersonknapp, 2019-10-10:
## Description

We have encountered deadlock conditions in using ROS2 services. This should never happen because of the core code. Users will always be able to build deadlocks, but the core shouldn't allow for it to happen.

## Documents

Needs more information

## Completion Criteria

ROS2 Services can't deadlock under normal usage conditions

emersonknapp, 2019-10-10:
Iceboxing until we have more information

thomas-moulard, 2019-10-14:
Camera is the poc for this one.

raghaprasad, 2020-01-21:
I don't have code samples where we encountered this, but here is the context: 

Cameron was attempting to build a prototype performance tests and wrote some python code that:
1. sets up a parameter server to vend values 
2. invoke the parameter server in a tight loop to read values 

We were noticing that after 3 or 4 retries the parameter server would deadlock and become unresponsive until we killed everything and restarted. 
Also note that this test was being done on ROS2 Dashing before it was launched and may have since been fixed.

emersonknapp, 2020-02-05:
I am going to recommend we close this issue as no longer a problem. Lots of changes have been made since Crystal and this has not been re-reported. 

The following test nodes reproduce the described scenario as closely as I could determine, and has no issues sucessfully calling the GetParameters service in an unregulated loop, achieving ~1 KHz, in repeated tests

```
# parameter_server.py
import rclpy
from rclpy.node import Node
from rclpy.parameter import Parameter

class ParameterHaver(Node):
    def __init__(self):
        super().__init__(
            'parameter_haver',
            parameter_overrides=[
                Parameter('foo', value=True),
                Parameter('bar', value=12),
                Parameter('baz', value='quux'),
            ],
            automatically_declare_parameters_from_overrides=True,
        )

def main(args=None):
    rclpy.init(args=args)
    node = ParameterHaver()
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

```
# parmeter_getter.py
from rcl_interfaces.srv import GetParameters
import rclpy

def main(args=None):
    rclpy.init(args=args)
    node = rclpy.create_node('parameter_getter')
    getter_client = node.create_client(GetParameters, 'parameter_haver/get_parameters')
    while not getter_client.wait_for_service(timeout_sec=1.0):
        print('service not available, waiting again...')

    success_count = 0
    failure_count = 0

    def check_rate():
        nonlocal success_count
        nonlocal failure_count
        node.get_logger().info('Last 1s: {} successes, {} failures'.format(
            success_count, failure_count))
        success_count = 0
        failure_count = 0

    node.create_timer(1.0, check_rate)

    while rclpy.ok():
        req = GetParameters.Request(names=['foo', 'bar', 'baz'])
        future = getter_client.call_async(req)
        rclpy.spin_until_future_complete(node, future)
        if future.result() is not None:
            # node.get_logger().info('Result of service: %s' % str(future.result()))
            success_count += 1
        else:
            node.get_logger().error('Exception while calling service: %r' % future.exception())
            failure_count += 1

    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

raghaprasad, 2020-02-05:
Thanks for testing this @emersonknapp 
Before we close the ticket can we extend this to include a setParam call as well? 
I searched the documentation and could not find a limit on set / get frequency.

emersonknapp, 2020-02-05:
I've run this with a SetParam followed by GetParam (to ensure that values were set) in a tight loop and am not seeing any issues. There would be no limit on frequency of calls, it should only be limited by the time it takes to actually perform the operations. My computer is achieving  ~1K service roundtrips per second in both test cases.

emersonknapp, 2020-02-05:
If I run 2 getters simultaneously I get slightly slower rate on each but surprisingly nowhere near half. Suggests that most time is taken up in the networking layer. Not relevant, but interesting.

thomas-moulard, 2020-02-05:
I think this bug has been fixed. @raghaprasad feel free to re-open if you disagree!

closed 2020-02-05 by thomas-moulard