ros-tooling/aws-roadmap #65: As a ROS2 developer, I want a standard way to send and receive runtime metrics, so that I can start to create tools around it

emersonknapp, 2019-10-11:
## Description

We want to define one standard channel for runtime metric information for ROS2. This channel can contain topic statistics, system metrics, and user-defined custom information.

To support those workflows we need to provide a core to easily report and listen to this information.

## Documents

## Completion Criteria

* [ ] A message type has been defined for runtime metric information
* [ ] A standard topic(s) has been chosen for runtime metric information
* [ ] A library/tool for aggregating and publishing these messages has been released for C++ and Python

emersonknapp, 2019-10-22:
Iceboxing this library creation until we have assessed the System and Metrics messages nodes and decide to open the collection API for use.

dabonnie, 2020-01-23:
Closing this out as the items are duplicates of other issues.

closed 2020-01-23 by dabonnie