ros-tooling/aws-roadmap #67: Trial and measure performance of proposed approach (without internal code)

emersonknapp, 2019-10-11:


dabonnie, 2019-10-18:
Investigation results of compressing a written rosbag2 file.

Reference **13GB** file provided by the RoboMaker Team. Results are tabulated below:

|  Compression Algorithm |  Size (GB) | Ratio  | Time Taken  | 
|---|---|---|---|
| bzip2 (levels 1 & 9, no change)  | 1.8  | 7.2  | 8m42.868s  |  
| lz4 fast (default)  |  5.8 | 2.2  | 1m24.922s  |  
|  lz4 best (level 9) | 3.2  | 4.1  | 34m54.570s  |  
| lz4 mid (level 4)  |  4.4 | 3.0  | 3m32.974s  |  
| Zstandard (default=3)  |  2.6 |  5.0 |  1m50.756s |  
| Zstandard (level 7)  | 2.4  |  5.4 | 3m35.970s  |  
| Zstandard (level 1)  |  2.6 | 5.0  | 1m26.618s  |  
| Snappy  | 4.3   | 3.0  | 1m22.894s  |  

[Internal Amazon Doc Link](https://quip-amazon.com/39RdAmVZIRLo)

dabonnie, 2019-10-18:
Next steps: trial and measure performance of compressing individual messages.

closed 2019-10-18 by emersonknapp