ros-tooling/aws-roadmap #75: As a ROS2 user, I can depend easily on wjwwood/serial so that my ROS2 code can communicate with rs232 peripherals

emersonknapp, 2019-10-12:
## Description

REMOVE ME: A short description of this large feature / feature set. An epic should still be well scoped and have a clear concept of where "finished" is.

## Documents

REMOVE ME: Link or include any designs that inform this Epic's completion criteria.

## Completion Criteria

REMOVE ME: Instead of creating a list here, start creating issues under this epic, this will be the TO-DO list for this Epic. When all children are done, this Epic is done. That list may fluctuate as requirements are discovered/negotiated.

zmichaels11, 2019-11-11:
Rewriting CMakeLists.txt to use `ament`

zmichaels11, 2019-11-11:
Marked `milestone` as `Sprint #6` since this was started late on Thursday, Nov 07 (during sprint 6)

zmichaels11, 2019-11-11:
### To Do:
- [ ] Convert project build from `catkin` to `ament`
- [ ] Convert test build from `catkin` to `ament`
- [ ] Run linters with `colcon test`
- [ ] Run tests with `colcon test`

zmichaels11, 2019-11-11:
After rewriting `CMakeLists.txt` to `ament`, this package fails every linter with several changes required from `uncrustify` and `cpplint`

`-Werror=sign-compare` and `-Werror=unused-but-set-variable` caused multiple errors in `list_ports_linux.cc`

This task might need to be repointed due to amount of rewrite required.

thomas-moulard, 2019-11-11:
Could we just disable the linters?

zmichaels11, 2019-11-18:
`'boost/bind.hpp' file not found` on macOS GH Action; testing on local machine

thomas-moulard, 2019-11-18:
Is it in a test, boost is only listed as a test_depend, that may be the problem here.

zmichaels11, 2019-11-18:
It is listed in the test `tests/unix_serial_tests.cc`

thomas-moulard, 2019-11-18:
Then it may be a problem with my `rosdep` invocation in the CI, I'd check and see if boost is getting installed or not. My guess is that it's not pulled properly (maybe rosdep does not include test dep by default, that would make sense?), and it just happens to work in most cases because Ubuntu is probably pre-loaded with Boost, but not the OS X worker? Also, you're unlikely to be able to repro locally if you have boost installed.

zmichaels11, 2019-11-21:
Boost is installed on OSX, however its CMake files are being skipped.

* The ROS1 code for serial also targets hydro. It might be a good idea to spin off a user story requesting it to be upgraded to a more recent ROS1.

zmichaels11, 2019-11-21:
Boost was safely removed as a dependency.
Build and test for OSX + Ubuntu 18.04: [image]

zmichaels11, 2020-01-03:
@zmichaels11 add user story to bring serial to updated ros1 distro

piraka9011, 2020-01-09:
It looks like bringing this package up to speed to both ROS1 and 2 will require more refactoring and more use stories. 
We should decide as a team whether we want to commit to supporting this further or not maintaining it.

piraka9011, 2020-01-13:
Closing since we won't be supporting this.

zmichaels11, 2020-05-28:
Not planing to do anymore

closed 2020-05-28 by zmichaels11