ros-tooling/aws-roadmap #76: Python support intra-process communication

emersonknapp, 2019-10-12:
## Description

"As a ROS 2 Python user, I want rclpy to support IPC
communication so that ROS 2 in Python reaches an
acceptable level of performance"

rclcpp ROS2 clients have access to the IntraProcessManager, which allows them to have zero-copy message passing within a single process.

Python users do not have access to this feature. They need it

## Documents

## Completion Criteria

mjcarroll, 2019-10-15:
Just saw that you mentioned this from the zero-copy side.  While zero-copy may be useful here, it's currently limited to POD messages and needs a middleware that implements the zero-copy (likely through shared memory).

I imagine a more complete implementation would include re-implementing an IntraProcessManager in Python.

emersonknapp, 2019-10-15:
Thanks for the info - I wasn't sure how related it was.

As for 
> re-implementing an IntraProcessManager in Python

We're worried about this trend in ROS2 - it seems like one of the major problems in ROS1 that the `rcl` architecture was intended to avoid (do we need to reimplement it in `rclc`, `rcljava`, `rclrust`, etc?). Ideally we can kick off a more serious conversation about a more universal solution, maybe something that lives in `rcl` and can be exposed to language clients thinly.

mjcarroll, 2019-10-15:
I think it depends on if intra-process is considered a "language specific feature" or not per https://index.ros.org//doc/ros2/Concepts/ROS-2-Client-Libraries/#id4

There is probably an argument to each side here, as specific intraprocess mechanisms may be used in one client implementation versus another.  An `rcl`-level implementation may also be considered, though.

thomas-moulard, 2020-01-14:
We are assembling the requirements for this project now. Waiting on feedback.

emersonknapp, 2020-07-24:
Closing on this roadmap, will discuss elsewhere

closed 2020-07-24 by emersonknapp