ros-tooling/aws-roadmap #77: As a ROS user, I want to prevent multiple nodes from having the same name

emersonknapp, 2019-10-12:
## Description

"As a ROS 2 user, I want to prevent multiple nodes
from having the same name so that I avoid hard to
debug issues and potentially unsafe behaviors"

An existing "request for design" is here, but the future is very uncertain
https://github.com/ros2/design/issues/187

thomas-moulard, 2020-01-14:
Proposals:

- modify ROS core packages to log a warning if multiple nodes with the same name are present on the network
- modify ROS2 launch to emit a warning if multiple nodes with the same name are present in a particular launch configuration
- modify `ros2 node` verb to print output in a way that nodes with the same node can be differentiated (host ip + pid?). `node` -> `node[127.0.0.1/1234]`

thomas-moulard, 2020-01-15:
The original issue is this one: https://github.com/ros2/ros2cli/issues/279

emersonknapp, 2020-02-24:
Resolved via spike report and new stories

closed 2020-02-24 by emersonknapp