ros-tooling/aws-roadmap #80: Adding required structs and methods to get a list of publishers or subscribers with their respective qos in rmw

jaisontj, 2019-10-14:
## Description
Add the required structs and functions needed by the RMW Layer to implement [this](https://github.com/ros2/rmw/issues/151) feature.

Namely, 
- functions to retrieve all publishers and subscriptions to a topic along with their QoS Policies
- structs to encapsulate a list of publisher/subscription name and respective QoS Policy

## Test Plan
* Unit test for new functions

## Documentation Plan
* Comments for every new structure and function

## Release Plan
- Submit PR to the [ros2/rmw](https://github.com/ros2/rmw)

## Acceptance Criteria
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

jaisontj, 2019-10-14:
[PR Submitted](https://github.com/ros2/rmw/pull/186)

prajakta-gokhale, 2019-10-28:
Keeping open until PR is merged.

closed 2020-01-21 by piraka9011