ros-tooling/aws-roadmap #83: ros2/rmw_fastrtps modify topic_cache to include publisher and subscriber QoS Policies

jaisontj, 2019-10-14:
## Description
As a step towards implementing [this](https://github.com/ros2/rmw/issues/151) feature, topic_cache needs to cache publisher and subscriber QoS Policies along with the topic name and types.

Relevant discussion about this is [here](https://github.com/ros2/rmw_fastrtps/issues/319).

## Test Plan
* Manual testing
* Unit test for `topic_cache`

## Documentation Plan
Comments for the functions need to be updated.

## Release Plan
Submit PR to [ros2/rmw_fastrtps](https://github.com/ros2/rmw_fastrtps)

## Acceptance Criteria
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed

Once all above items are checked, this story can be moved to done.

jaisontj, 2019-10-18:
[PR](https://github.com/ros2/rmw_fastrtps/pull/332)

emersonknapp, 2019-10-18:
Can you also reverse that link? "Connected to aws-roadmap#83" on the PR description?

mm318, 2020-01-14:
This has been superseded by https://github.com/ros-security/aws-roadmap/issues/84.

closed 2020-01-14 by mm318