ros-tooling/aws-roadmap #85: ros2/rclpy Create and implement functions to retrieve publishers and subscribers (with their QoS Policies) to a topic

jaisontj, 2019-10-14:
## Description
- Implement the equivalent of the functions defined [here](https://github.com/ros-security/aws-roadmap/issues/80) in the `_rclpy.c` bridge.
- Add a helper method to convert the retrieved list to a PythonObject (List of Dictionaries)

## Test Plan
* Manual testing
* Unit test for new functions

## Documentation Plan
Comments for modified and new functions

## Release Plan
Submit PR to [ros2/rclpy](https://github.com/ros2/rclpy/)

## Acceptance Criteria
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed

Once all above items are checked, this story can be moved to done.

jaisontj, 2019-10-29:
[PR](https://github.com/ros2/rclpy/pull/454)

mm318, 2019-12-11:
An issue to track the `rclcpp` analogue of this has been created here https://github.com/ros2/rclcpp/issues/943. A corresponding issue hasn't been created under aws-roadmap yet, because it hasn't been decided yet who will be implementing it.

thomas-moulard, 2020-02-03:
Punted from sprint #11 to sprint #12 on 2020-02-03

closed 2020-02-14 by thomas-moulard