ros-tooling/aws-roadmap #86: ros2/ros2cli Display all publishers and subscribers with their QoS Policies for a given topic

jaisontj, 2019-10-14:
## Description
- Call the respective underlying functions to retrieve all publishers and subscribers (with their QoS Policies) for a topic.
- Display data

```
$ ros2 topic info rosout

Topic: rosout
Publisher count: 1

Publisher Name: _ros2cli_daemon_0
QoS Profile: 
    Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
    Durability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
    Lifespan: 1000000000 nanoseconds
    Deadline: 2000000000 nanoseconds
    Liveness: RMW_QOS_POLICY_LIVELINESS_MANUAL_BY_TOPIC
    Liveness Lease Duration: 5 nanoseconds

Subscriber count: 0
```

## Test Plan
* Manual testing
* End to end integration test calling CLI entrypoint

## Documentation Plan
- Comments for modified and new functions
- add it to the cli help output

## Release Plan
Submit PR to [ros2/ros2cli](https://github.com/ros2/ros2cli/)

## Acceptance Criteria
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

jaisontj, 2019-10-24:
[PR](https://github.com/ros2/ros2cli/pull/385)

thomas-moulard, 2020-02-03:
Punted from sprint #11 to sprint #12 on 2020-02-03

closed 2020-02-28 by thomas-moulard