ros-tooling/aws-roadmap #89: As a rviz2 user, I want to be able to visualize compressed images so that I can stream camera images from my robot efficiently

emersonknapp, 2019-10-16:
## Description

"As a rviz2 user, I want to be able to visualize
compressed images so that I can stream camera
images from my robot efficiently"

REMOVE ME: A short description of this large feature / feature set. An epic should still be well scoped and have a clear concept of where "finished" is.

## Documents

REMOVE ME: Link or include any designs that inform this Epic's completion criteria.

## Completion Criteria

REMOVE ME: Instead of creating a list here, start creating issues under this epic, this will be the TO-DO list for this Epic. When all children are done, this Epic is done. That list may fluctuate as requirements are discovered/negotiated.

thomas-moulard, 2019-11-07:
This work has already been done by Open Robotics. We need to make sure that we can:
a/ run image_transport in ROS 2
b/ display a compressed camera data stream in rviz2 successfully

zmichaels11, 2020-02-04:
I found [this](https://github.com/ros-perception/image_common/wiki/ROS2-Migration) which states it has been migrated to ROS2, I'll look more into if this is usable.

zmichaels11, 2020-02-06:
There's a `ros2` branch of `image_transport_plugins` that includes PNG and JPEG compression.
Last commit was 14 months ago

zmichaels11, 2020-02-06:
`ros2 run image_transport list_transports` works under Dashing.
```bash
sudo apt install ros-dashing-compressed-image-transport
sudo apt install ros-dashing-theora-image-transport
ros2 run image_transport list_transports
```
output:
```bash
Declared transports:
image_transport/compressed
image_transport/raw
image_transport/theora

Details:
----------
"image_transport/compressed"
 - Provided by package: compressed_image_transport
 - Publisher: 
      This plugin publishes a CompressedImage using either JPEG or PNG compression.
    
 - Subscriber: 
      This plugin decompresses a CompressedImage topic.
    
----------
"image_transport/raw"
 - Provided by package: image_transport
 - Publisher: 
      This is the default publisher. It publishes the Image as-is on the base topic.
    
 - Subscriber: 
      This is the default pass-through subscriber for topics of type sensor_msgs/Image.
    
----------
"image_transport/theora"
 - Provided by package: theora_image_transport
 - Publisher: 
      This plugin publishes a video packet stream encoded using Theora.
    
 - Subscriber: 
      This plugin decodes a video packet stream encoded using Theora.
```

zmichaels11, 2020-02-11:
* The [tutorial](https://github.com/ros-perception/image_common/tree/ros2/image_transport) has not been ported to ROS 2
* Documentation published on differences does not match; `ImageTransport` requires a `shared_ptr<Node>`
* ament dependencies doesn't properly pull in OpenCV on Eloquent
* opencv_bridge states it doesn't support `image_transport` yet in a [comment block](https://github.com/ros-perception/vision_opencv/blob/a7576d528891a154df3689b02ba5a6399ffaa41c/cv_bridge/include/cv_bridge/cv_bridge.h#L332) 
* [Rewrote demo in ROS 2](https://github.com/zmichaels11/image_transport_demo)
* Wrote steps in `README` how to use `republish` to compress images (tested on Eloquent)
* rviz2 is capable of subscribing to compressed image topics, however camera and image display don't receive the topics
* image_transport republish does transcode images successfully

closed 2020-02-14 by thomas-moulard