ros-tooling/aws-roadmap #93: As a ROS 2 user, I want to be able to call services inside callbacks so that I can write complex callbacks easily

emersonknapp, 2019-10-16:
## Description

"As a ROS 2 user, I want to be able to call services
inside callbacks so that I can write complex
callbacks easily"

REMOVE ME: Fill out this and all below sections, and remove the PROMPT questions for readability.

PROMPT: A concise, concrete, finishable, description of the feature to be built.

## Test Plan

PROMPT: What tests should be created to support this feature?
* Manual testing
* Unit test for new classes
* End to end integration test calling CLI entrypoint
* ...

## Documentation Plan

PROMPT: What documentation updates need to be made to support this feature? e.g.
* README.md update
* wiki update
* design doc PR
* ...

## Release Plan

PROMPT: What release work needs to be done to finish this feature? e.g.
* check in PR
* bloom release

## Acceptance Criteria

PROMPT: Clearly defined checklist of things that need to be done to call this task complete. Can include items from all above sections. 
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

PROMPT: Add here links to any relevant document e.g. design (and make sure everyone can access the docs linked here!):*
* [FIXME](http://...)

thomas-moulard, 2020-01-14:
Waiting for repro code before investigating this.

emersonknapp, 2020-02-04:
I believe this issue may be in reference to this line https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/client.py#L69 

The _synchronous_ service call API is not recommended in callbacks because it may block indefinitely and eat up your callback threads, hence the deadlock.

A potential solution to closing this issue is a demo/example that uses `client.call_async` in a callback to show the recommended way to perform these operations.

emersonknapp, 2020-02-05:
Relevant conversation https://answers.ros.org/question/343279/ros2-how-to-implement-a-sync-service-client-in-a-node/

thomas-moulard, 2020-02-05:
@mm318 Seems like Emerson has some ideas about what needs to be done there. Would you mind sync-ing with him?

mm318, 2020-02-10:
Is this an issue with `rclpy` only? What about `rclcpp`? Or a system with a mix of `rclcpp`/`rclpy`?

mm318, 2020-02-12:
Created spike for this milestone at https://github.com/ros-tooling/aws-roadmap/issues/184.

mm318, 2020-02-13:
The discussion following the spike reached the following conclusion on possible resolution:
- Better education on ROS2 programming best practices (through documentation, tutorials, etc.)
  - Currently, there's only some obscure mentions of the issue on ROS Answers ([here](https://answers.ros.org/question/343279/ros2-how-to-implement-a-sync-service-client-in-a-node/) and [here](https://answers.ros.org/question/302037/ros2-how-to-call-a-service-from-the-callback-function-of-a-subscriber/))
- Address the `rclpy` API:
  - Deprecate the synchronous `Client.call()` API
    - Emit a warning whenever `Client.call()` is used
  - Remove the synchronous `Client.call()` API altogether (even `rclcpp` does not have an equivalent)

Immediate next step is to start a discourse post regarding this issue.

mm318, 2020-02-14:
Discourse post was created here: https://discourse.ros.org/t/synchronous-request-to-service-in-callback-results-in-deadlock/12767

closed 2020-02-14 by thomas-moulard