ros-tooling/aws-roadmap #97: As a ROSBag2 developer, I can create new Reader types so that I can extend the functionality of the ROSBag2 Readeer.

piraka9011, 2019-10-18:
## Description

The current implementation of the `SequentialReader` in ROSBag2 is a single class that reads the next message from a storage plugin. 
If we want to extend ROSBag2 with more features, we need to make an interface for a ROSBag2 developer to extend with the necessary features.

There will be a new `IReader` class used as an interface. This will be used to _compose_ a `SequentialReader`, which reads messages from a storage plugin in sequence and a `MultiFileReader` which can read from multiple split bag files.

## Test Plan

* Manual testing
* Unit test for the `SequentialReader` and the `MultiFileReader`
Integration tests are part of another user story.

## Documentation Plan

C++ Doxygen.

## Release Plan

Merge into the `master` branch of `ros2/rosbag2`.

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* Most changes can be found in [aws-ros-dev/splitting-reader-changes](https://github.com/aws-ros-dev/rosbag2/tree/splitting/reader-changes)

piraka9011, 2019-10-21:
Work being done [here](https://github.com/aws-ros-dev/rosbag2/tree/reader-interface)

piraka9011, 2019-10-22:
The PR against rosbag2 master is [here](https://github.com/ros2/rosbag2/pull/191)

prajakta-gokhale, 2019-11-18:
https://github.com/ros2/rosbag2/pull/191 is merged, this feature is in master.

closed 2019-11-18 by prajakta-gokhale