ros-tooling/aws-roadmap #99: ros2cli test fails to shutdown subprocesses on Windows

emersonknapp, 2019-10-21:
## Description
https://github.com/ros2/ros2cli/pull/283 added some tests for ros2topic, but the subprocess actions in it are not working properly on Windows. The test has been disabled on that platform for now.

A new comprehensive test framework is being introduced in https://github.com/ros2/ros2cli/pull/304. Instead of fixing the ad-hoc test for ros2topic, refactor its QoS test to follow those patterns, once they are merged.

## Expected Behavior
ros2topic test_echo_pub tests to run and pass on Windows.

## Actual Behavior
The first test raises an exception on introducing the `SIGINT` signal to the subprocess running the CLI. All subsequent attempts by the pytest process to start a subprocess fail to even start the subprocess. 

These tests are currently disabled on Windows.

## To Reproduce

1. `colcon test --packages-select ros2topic`

## System (please complete the following information)
 - OS: Windows 10
 - ROS2 Distro: master

jacobperron, 2019-10-22:
See https://github.com/ros2/ros2cli/issues/375

hidmic, 2019-10-22:
Well, Windows processes do not react to SIGINT just as you'd expect pretty much everywhere else. You may check ROS 2 launch [`ExecuteProcess` action implementation](https://github.com/ros2/launch/blob/master/launch/launch/actions/execute_process.py#L386-L394) to check how this inconvenience was circumvented in the past. In any case, as @jacobperron said, we should probably move forward with https://github.com/ros2/ros2cli/issues/375. That'll address this issue as a side effect.

prajakta-gokhale, 2019-11-06:
The tracked issue https://github.com/ros2/ros2cli/issues/375 has been closed and the fix is in https://github.com/ros2/ros2cli/pull/377.

closed 2019-11-08 by prajakta-gokhale