ros-tooling/aws-roadmap #102: As a ROS2 user, I want rosbag2 to record my custom message types so that I can inspect and play back my application

emersonknapp, 2019-10-22:
## Description

Tracking issue for https://github.com/ros2/rosbag2/issues/193

## Documents

## Completion Criteria

piraka9011, 2020-01-09:
Closing since this seems to be a larger ROS2 issue not related to rosbag2.

thomas-moulard, 2020-01-09:
This is still a problem for rosbag2 user, so we'll need to fix it one way or another.

emersonknapp, 2020-01-09:
This isn't an issue as it is currently written. If you want to keep this open, you'll need to specify that it only applies to remote computers being listened to over the network

piraka9011, 2020-01-09:
Moreover, I don't think this is a rosbag2 specific issue. It makes sense that rosbag2 can read messages supported/compiled on the system it is running on as well as providing support for different serialization formats.

Outside of [abstracting the subscription interface](https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/generic_subscription.hpp#L29) in rosbag2, I'm not sure what else could be done to support this use case.

emersonknapp, 2020-02-28:
Closing this issue as irrelevant (as written)