ros-tooling/aws-roadmap #112: Measure CPU and Memory Per Process

dabonnie, 2019-10-31:
## Description

Using the same tools developed to measure system cpu and memory, measure these metrics per process via a composable node. 

## Test Plan

PROMPT: What tests should be created to support this feature?
* Manual testing
* Unit test for new classes
* End to end integration test calling CLI entrypoint

## Documentation Plan

* README.md update

## Release Plan

PROMPT: What release work needs to be done to finish this feature? e.g.
* check in PR

## Acceptance Criteria

PROMPT: Clearly defined checklist of things that need to be done to call this task complete. Can include items from all above sections. 
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [Design Doc](https://quip-amazon.com/KzzfAUQeRkVY/Design-Doc-ROS2-Runtime-Metrics-System-Metrics)

zmichaels11, 2019-12-03:
TODO: add an example in the README

dabonnie, 2019-12-03:
In case measuring the CPU / Memory for a process (not system) is more complex than anticipated (different proc file contents) then we will split the measurement and composeable portions into separate stories.

dabonnie, 2019-12-04:
Filed https://github.com/ros-security/aws-roadmap/issues/139

dabonnie, 2019-12-04:
Closing in favor of the smaller user stories.

closed 2019-12-04 by dabonnie