ros-tooling/aws-roadmap #121: As a rosbag2 developer, I want a compression interface so I can develop compression plugins

piraka9011, 2019-11-15:
## Description

A new compression interface class needs to be added to `rosbag2` so that new compression plugins can be developed for compressing messages and files.
The interface needs the following methods (at least):
* [ ] Compress File
* [ ] Compress SerializedBagMessage
* [ ] Get a compression ID

## Test Plan

3 Unit tests:
* File compression is called if file compression is in metadata.
* Message compression is called if message compression is in metadata.
* No compression is called is no compression is in metadata.

## Documentation Plan

Docstrings

## Release Plan

* Merge in to master branch of rosbag2

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [FIXME](http://...)

zmichaels11, 2019-12-06:
@piraka9011 can you list any related PRs to this user story?

piraka9011, 2019-12-06:
Related PR: https://github.com/ros2/rosbag2/pull/219

zmichaels11, 2019-12-06:
Did you mean ros2/rosbag2#218 ?

piraka9011, 2019-12-06:
Yes, thanks :)

closed 2019-12-06 by zmichaels11