ros-tooling/aws-roadmap #124: As a rosbag2 user, I want to specify my compression format and method so that I can compress my bags

piraka9011, 2019-11-15:
## Description

The de/compression features need to be exposed to a rosbag2 user through the CLI. 
Two flags need to be added to the `record` verb:
* [ ] `-c --compression-method` (FILE, MESSAGE)
* [ ] `-a --compression-algorithm` (ZSTD)

## Test Plan

Argparse unit test that checks the proper flags are propagated to the `rosbag2_transport` layer.

## Documentation Plan

Update argparse epilog/help docstrings.

## Release Plan

* Merge into master branch of `rosbag2`

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [FIXME](http://...)

zmichaels11, 2020-01-21:
- [ ] Add `SequentialCompressionWriter` PR: https://github.com/ros2/rosbag2/pull/260

zmichaels11, 2020-01-22:
- [ ] PR: https://github.com/ros2/rosbag2/pull/263

thomas-moulard, 2020-02-03:
Punted from sprint #11 to sprint #12 on 2020-02-03

closed 2020-02-14 by thomas-moulard