ros-tooling/aws-roadmap #125: Issue to track progress of Epic #115

jaisontj, 2019-11-16:
This EPIC can be tracked via [this](https://github.com/ros2/ros2/issues/822) issue. 

## List of PRs to be tracked towards this: 
- [ ] [ros2/rmw#193](https://github.com/ros2/rmw/pull/193/files)
- [ ] [ros2/rcl#535](https://github.com/ros2/rcl/pull/535)
- [ ] [ros2/rclpy#459](https://github.com/ros2/rclpy/pull/459)
- [ ] [ros2/rclcpp#924](https://github.com/ros2/rclcpp/pull/924/files)
- [ ] [ros2/demos#416](https://github.com/ros2/demos/pull/416)
- [ ] [ros2/rmw_opensplice#294](https://github.com/ros2/rmw_opensplice/pull/294/files)

thomas-moulard, 2020-02-03:
Punted from sprint #11 to sprint #12 on 2020-02-03

mm318, 2020-03-04:
This ticket is actually not the epic. The epic is at https://github.com/ros-tooling/aws-roadmap/issues/115.

mm318, 2020-03-30:
Ideally, there are sprint points assigned to this ticket, but it is quite hard to accurately point.

If this story was about the initial implementation of incompatible QoS callbacks, it might be easy to point. But this story is about pushing the pull requests through, and it largely depends on the amount and type of upstream maintainer/reviewer feedback, which is highly variable.

emersonknapp, 2020-03-30:
This is the whole purpose of "relative complexity" estimates instead of "absolute time" ones. `rmw` API changes should, based on our experience, have a higher relative complexity because we know they take a lot of care and consideration, whereas even if rclcpp may be more code, it's probably a lower relative complexity because by the time you're reviewing it, it's probably noncontroversial, there's much less uncertainty.

emersonknapp, 2020-03-30:
And this ticket wouldn't have points, it's meant to track an Epic

closed 2020-04-10 by emersonknapp