ros-tooling/aws-roadmap #132: Add ros1 support to ros2 cross-compile tool

prajakta-gokhale, 2019-11-19:
## Description

Currently the [cross-compile](https://github.com/ros-tooling/cross_compile) tool cross compiles ROS 2 source files. We want the same tool to support cross compilation of ROS 1 source files.

## Test Plan

* Manual testing by cross compiling 1 RoboMaker SampleApplication for supported target platforms
* New unit test if necessary
* End to end integration test for ROS 1

## Documentation Plan

* README.md update to document new feature

## Release Plan

* Merge PR into `master`
* PIP release

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [cross-compile repo](https://github.com/ros-tooling/cross_compile)
* [HelloWorld sample app](https://github.com/aws-robotics/aws-robomaker-sample-application-helloworld)

emersonknapp, 2019-12-10:
General usability changes going through clearing the way for ROS1 support
* https://github.com/ros-tooling/cross_compile/pull/71
* https://github.com/ros-tooling/cross_compile/pull/72
* https://github.com/ros-tooling/cross_compile/pull/75
* https://github.com/ros-tooling/cross_compile/pull/74
* https://github.com/ros-tooling/cross_compile/pull/76

prajakta-gokhale, 2019-12-13:
ROS1 support

* [ ] https://github.com/ros-tooling/cross_compile/pull/90

prajakta-gokhale, 2019-12-19:
Currently developing GitHub Actions workflow to automate the e2e testing.

prajakta-gokhale, 2020-01-03:
Bloom release:

* [ ] Dashing - https://github.com/ros/rosdistro/pull/23394
* [ ] Eloquent - https://github.com/ros/rosdistro/pull/23395

Since cross_compile is a ROS 2 package, we won't be bloom releasing it into Kinetic and Melodic right now. Instead, we want to convert cross_compile into a pure Python package and release it via `pip` so it is easier to maintain and release, and the installation steps remain same for all supported ROS distros (https://github.com/ros-tooling/cross_compile/issues/102).

closed 2020-01-06 by piraka9011