ros-tooling/aws-roadmap #136: As a rosbag2 developer, I want GH Actions CI to run on rosbag2_bag_v2 periodically to verify that upstream changes did not break the build.

zmichaels11, 2019-11-25:
## Description
Run GH Actions on `rosbag2_bag_v2` on push to master, PR, and periodically.
Follow up on this PR: https://github.com/ros2/rosbag2_bag_v2/pull/10

## Test Plan
* Manual Testing
  * Verify no errors in build logs

## Documentation Plan
* Update `rosbag2_bag_v2` README.md with build badge

## Release Plan
* check in PR

## Acceptance Criteria
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources
* Open PR: https://github.com/ros2/rosbag2_bag_v2/pull/10

zmichaels11, 2020-01-02:
PR for Linters: https://github.com/ros2/rosbag2_bag_v2/pull/12

piraka9011, 2020-01-06:
@zmichaels11 move to rosbag2_bag_v2 repo and close here.

zmichaels11, 2020-01-06:
Moving to `rosbag2_bag_v2` here: https://github.com/ros2/rosbag2_bag_v2/issues/14

closed 2020-01-06 by zmichaels11