ros-tooling/aws-roadmap #149: ROS2 System Metrics Launch File

dabonnie, 2019-12-13:
## Description

Create a launch file to run the system metrics system cpu and system memory nodes.
 
## Test Plan

PROMPT (REMOVE-ME): What tests should be created to support this feature?

* Manual testing
* End to end integration test calling launch file

## Documentation Plan

* README.md update

## Release Plan

* check in PR

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

prajakta-gokhale, 2020-01-08:
* [ ] Adding launch file: https://github.com/ros-tooling/system_metrics_collector/pull/63

dabonnie, 2020-01-08:
Filed https://github.com/ros-security/aws-roadmap/issues/162 to capture configuration of the nodes in the launch file.

closed 2020-01-21 by piraka9011