ros-tooling/aws-roadmap #150: As a rosbag2 user, I want to know how much system resources rosbag2 is using so I can generate performance benchmarks

piraka9011, 2019-12-17:
## Description

Using the [`ROS2_Performance_Testing_Tools`](https://code.amazon.com/packages/ROS2_Performance_Testing_Tools/trees/mainline/--/) package, we can obtain system level metrics such as memory and CPU usage. 
We should use this tool to get the following metrics and generate reports for the simulation team about how rosbag2 performs:

* CPU usage
* Memory usage

The measurements and report should be generated via a script and stored in a specific directory to compare against previous performance runs.

## Test Plan

TODO(piraka9011): Are any tests necessary for using this? Do we need new tests for scripts.

## Documentation Plan

Quip doc explaining how to generate the reports

## Release Plan

This does not have a release candidate.

## Acceptance Criteria

* [ ] Developed a script that can automatically generate CSV files and graphs of aforementioned metrics
* [ ] Documented workflow in quip doc

Once all above items are checked, this story can be moved to done.

## Resources

* TODO: [Quip Doc]()
* [ROS2_Performance_Testing_Tools](https://code.amazon.com/packages/ROS2_Performance_Testing_Tools/trees/mainline/--/) using `plot_system_stats.py` and `log_system_stats.py`

zmichaels11, 2019-12-20:
I think this is actually a spike

piraka9011, 2020-01-10:
Consider using System Metrics Collector instead.
Use the sample apps to test.

piraka9011, 2020-02-03:
A PR in rosbag2 with a performance test proposal was opened [here](https://github.com/ros2/rosbag2/pull/244).

dabonnie, 2020-02-04:
@zmichaels11 @piraka9011 

>Dropped messages (in rosbag2 and not in DDS layer)

Is this currently possible in rosbag2?

dabonnie, 2020-02-12:
Note: ros2 message compression was removed from the test matrix because it is not yet implemented

closed 2020-02-28 by thomas-moulard