ros-tooling/aws-roadmap #151: Demonstrate with PoC how to log performance metrics in rosbag2

piraka9011, 2019-12-17:
## Description

We need to determine how to log metrics to the CLI and (if possible within time frame) log to a CSV file.
Performance metrics and in which rosbag2 packagesto calculate & log them should be outlined in a quip doc.

## Spike Timeboxing

2 Days

## Spike Outcome

 * List of `rosbag2` packages where certain metrics can be tracked
 * A PoC of logging to CLI/file and reading in results. 

## Goals

* [ ] Identified which rosbag2 packages are used to measure metrics
* [ ] Developed a PoC that logs metrics to CLI
* [ ] Document metrics and get initial feedback from simulation team

zmichaels11, 2020-01-02:
Is this related to https://github.com/ros-security/aws-roadmap/issues/150 ?

piraka9011, 2020-02-14:
Resolved by #181

closed 2020-02-14 by piraka9011