ros-tooling/aws-roadmap #152: As a rosbag2 user, I want to know when my messages have been dropped so I can be informed if rosbag2 has recording my message

piraka9011, 2019-12-17:
## Description

The `rosbag2` writer should log via the CLI whenever a message (compressed or uncompressed) has been dropped and not written to the storage plugin. 

## Test Plan

* Unit test demonstrating writer logs the warning if the message wasn't written to the database (mocking the storage plugin).
* Unit test demonstrating writer logs the warning in a compression workflow (i.e. the compression flag was passed to the write).

## Documentation Plan

Code docstrings

## Release Plan

This will be part of a PoC branch on `aws-ros-dev/rosbag2`

## Acceptance Criteria

* [ ] Writer can log on dropped messages
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [FIXME](http://...)

emersonknapp, 2020-07-31:
Closing in favor of https://github.com/ros2/rosbag2/issues/295

emersonknapp, 2020-07-31:
Closing in favor of https://github.com/ros2/rosbag2/issues/477

closed 2020-07-31 by emersonknapp