ros-tooling/aws-roadmap #154: Rosbag2 Zstd compression as a plugin

zmichaels11, 2019-12-20:
## Description
As a Rosbag2 user, I want compression to be an optional requirement for Rosbag2 so that I can omit if if I do not need it or supply my own compression format.

To complete this task, the package `rosbag2_compression` should be organized similar to `rosbag2_storage`. This will require:
* `rosbag2_compression` split into `rosbag2_compression` (for the base interfaces) and `rosbag2_compression_default_plugins` (for the implementations)
* `ZstdCompressor` and `ZstdDecompressor` need to be exported as plugins
* Define `CompressionFactory` (see `StorageFactory` and `SerializationFormatConverterFactory` for how to)
* Update `SequentialReader` and `SequentialWriter` to accept `CompressionFactoryInterface` with a default parameter to the `CompressionFactory`
* Update `SequentialReader` to construct a `BaseDecompressorInterface` by invoking a plugin load from `CompressionFactory`
* Update `SequentialWriter` to construct a `BaseCompressorInterface` by invoking a plugin load from `CompressionFactory`

## Test Plan
* Unit tests for using `CompressionFactory` with `Writer`
* Unit tests for using `CompressionFactory` with `Reader`

## Documentation Plan
* N/A at this stage

## Release Plan
* check in PR

## Acceptance Criteria
* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources
* [SerializationFormatConverterFactory](https://github.com/ros2/rosbag2/blob/master/rosbag2/include/rosbag2/serialization_format_converter_factory.hpp)
* [SerializationFormatConverterFactoryInterface](https://github.com/ros2/rosbag2/blob/master/rosbag2/include/rosbag2/serialization_format_converter_factory_interface.hpp)
* [StorageFactory](https://github.com/ros2/rosbag2/blob/master/rosbag2_storage/include/rosbag2_storage/storage_factory.hpp)
* [StorageFactoryInterface](https://github.com/ros2/rosbag2/blob/master/rosbag2_storage/include/rosbag2_storage/storage_factory_interface.hpp)

zmichaels11, 2019-12-20:
CLI Needs to error out similar to how storage does if storage plugin is not available

zmichaels11, 2019-12-20:
Risk: Rosbag2 metapackage?

zmichaels11, 2019-12-20:
This UserStory should influence the CLI user story - CLI user story should be specific on the type of compression format

zmichaels11, 2019-12-20:
Add as resource link plugin management example

zmichaels11, 2019-12-20:
This *might* be more work due to how complex plugins are

emersonknapp, 2020-03-30:
Duplicate of https://github.com/ros-tooling/aws-roadmap/issues/192

closed 2020-03-30 by emersonknapp