ros-tooling/aws-roadmap #156: PeriodicCollectorNode can be dynamically started and stopped after running

dabonnie, 2019-12-31:
##Description

Allow each PeriodicCollectorNode to be dynamically started and stopped after running by using the [Trigger message](https://github.com/ros2/common_interfaces/blob/master/std_srvs/srv/Trigger.srv), similar to [cloudwatch metrics](https://github.com/aws-robotics/cloudwatchmetrics-ros2/blob/master/cloudwatch_metrics_collector/src/main.cpp#L108). 

Instead of hardcoding the entire trigger message as above, consider using the request metadata (message field) for specific actions, e.g., start, stop, status.

## Test Plan

* Manual testing start and stop functionality
* Unit tests for start and stop functionality

## Documentation Plan

* README.md documentation update
* Document class

## Release Plan

* check in PR

## Acceptance Criteria

PROMPT (REMOVE-ME): Clearly defined checklist of things that need to be done to call this task complete. Can include items from all above sections.

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

zmichaels11, 2020-01-03:
Lifecycle manager?

thomas-moulard, 2020-01-09:
See https://design.ros2.org/articles/node_lifecycle.html

dabonnie, 2020-01-09:
> See https://design.ros2.org/articles/node_lifecycle.html

Thanks, already there! It's exactly what we need here :-)

dabonnie, 2020-01-15:
Opened PR: https://github.com/ros-tooling/system_metrics_collector/pull/72

dabonnie, 2020-01-21:
Opened PR for documentation: https://github.com/ros-tooling/system_metrics_collector/pull/77

closed 2020-01-22 by dabonnie