ros-tooling/aws-roadmap #161: As a ROS2 user, I want to use a controller in a Gazebo simulation so I can test my control algorithms

piraka9011, 2020-01-07:
## Description

As part of the work of porting ros_control from ROS1 to ROS2, we need to develop a Gazebo plugin that can read/write to/from a simulated hardware interface.

This plugin will use the Gazebo C++ SDK to generate bindings between a URDF model and ros_control hardware interfaces. The plugin extends `gazebo::ModelPlugin`, implementing:

* [ ] `Load`
* [ ] `Update`
* [ ] `Reset`

## Test Plan

* Unit tests:
  * Update/Read from a Gazebo model
  * Write setpoint (ex. velocity, force) to Gazebo model joint
  * Model joints are intialized

## Documentation Plan

* Docstrings
* Update gazebo_ros_pkgs wiki

## Release Plan

* Merge in Foxy branch of `gazebo_ros_pkgs`

## Acceptance Criteria

* [ ] A Gazebo plugin package has been created in gazebo_ros_pkgs for ros_control
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* See [gazebo_ros_pkgs](https://github.com/ros-simulation/gazebo_ros_pkgs) for examples
* [Gazebo ROS2 Integration docs](http://gazebosim.org/tutorials?tut=ros2_overview&cat=connect_ros)
* [Gazebo ROS contribution guide](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-contribution-guide)

piraka9011, 2020-01-10:
Update with description that it is blocked on an external user story.

emersonknapp, 2020-04-16:
Closing as it seems that OSRF gazebo team will be taking this on, can reopen later if needed

closed 2020-04-16 by emersonknapp