ros-tooling/aws-roadmap #163: As a rosbag2 developer, I want to catch any regressions in compression so I can make sure my compression worklflow is stable

piraka9011, 2020-01-09:
## Description

After the completion of the rosbag2 compression feature, we need to add an end-to-end test for `rosbag2_compression` in `rosbag2_test`. 

This should test the following cases:

* [ ] Bad compression flags are passed
* [ ] Compression by file
* [ ] Compression by message

## Test Plan

* End to end tests for compression run

## Documentation Plan

* Docstrings

## Release Plan

* Merge into rosbag2 master

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

zmichaels11, 2020-01-22:
Note:
* No implementation of `rosbag2_compression` currently supports compress by message.
* `process_execution_helpers` in `rosbag2_test_common` does not report exit code. This will be required to write e2e tests on bad compression arguments.

emersonknapp, 2020-04-09:
Closed as no longer relevant

closed 2020-04-09 by emersonknapp