ros-tooling/aws-roadmap #165: Configure Composable Nodes

dabonnie, 2020-01-14:
## Description

To further the work of https://github.com/ros-tooling/system_metrics_collector/pull/67#discussion_r366103842, allow a user to provide the configuration of a node. The parameters are

 - node name
 - collection period
 - publish period
 - publishing topic

## Test Plan

* Manual testing
  * Verify the composable nodes can be launched using configured parameters

* README.md update
 * document supported parameters

## Release Plan

* check in PR

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* Depends on: https://github.com/ros-security/aws-roadmap/issues/141

mm318, 2020-01-17:
With https://github.com/ros-tooling/system_metrics_collector/pull/73, the composeable nodes are now configurable. I think this issue is done, and the rest of the configurability story should be here https://github.com/ros-security/aws-roadmap/issues/162.

piraka9011, 2020-01-31:
Closed in favor of #162

closed 2020-01-31 by piraka9011