ros-tooling/aws-roadmap #166: As a ROS2 developer, I can use ament_clang_tidy to correctly find lint issues with my ROS2 packages

mm318, 2020-01-14:
##  Description

`ament_clang_tidy` in its current state is unusable as it is unable to find many headers in the project (see https://github.com/ament/ament_lint/issues/203).

The goal of this story is to fix/enhance `ament_clang_tidy` to pass in pre-existing valid `compile_commands.json` files of a set of projects to `clang-tidy` for the linter to work correctly. `ament_clang_tidy` / `clang-tidy` linter will find lint issues and output them to a file, the screen, etc. Fixing the issues is out of the scope of this story.

## Test Plan

* Manually run `ament_clang_tidy` on some packages and confirm that the lint issues found make sense

## Documentation Plan

* Update the documentation [here](https://github.com/ament/ament_lint/blob/master/ament_clang_tidy/doc/index.rst) with instructions on how to generate `compile_commands.json` and run `ament_clang_tidy`

## Release Plan

* Merge in a PR into the [repo](https://github.com/ament/ament_lint)

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [StackOverflow answer to solving `clang-tidy` being unable to find header files](https://stackoverflow.com/a/39596822)

zmichaels11, 2020-01-17:
This might be a good OnCall task

piraka9011, 2020-01-31:
Consider using this: https://github.com/aws-robotics/rosbag-uploader-ros1/tree/master/.github/actions/clang-tidy

thomas-moulard, 2020-02-01:
@piraka9011 We should only use the ament one, if we follow your suggestions, we'll receive a different set of warnings, because the clang-tidy versions + config is not guaranteed to stay in sync.

emersonknapp, 2020-07-31:
Fixed in https://github.com/ament/ament_lint/pull/213

closed 2020-07-31 by emersonknapp