ros-tooling/aws-roadmap #167: Create System Metrics Examples

dabonnie, 2020-01-23:
## Description

Create example launch files in order to demonstrate how to configure, run, and instrument nodes with the system metrics collector composable nodes.

## Test Plan

* Manual testing
  * verify launch files start all nodes

## Documentation Plan

* README.md update
  * document example entry points, how to launch, and how to inspect
  * provide example output

## Release Plan

* check in PR

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [ROS2 Composition](https://index.ros.org/doc/ros2/Tutorials/Composition/)

dabonnie, 2020-01-23:
Opened PR to create example package: https://github.com/ros-tooling/system_metrics_collector/pull/81

emersonknapp, 2020-04-09:
This is done

closed 2020-04-09 by emersonknapp