ros-tooling/aws-roadmap #170: Colcon-bundle daily test failure sends an email to the oncall

emersonknapp, 2020-01-24:
## Description

Depends on https://github.com/ros-security/aws-roadmap/issues/169

When the end to end tests fail, page the on-call for the package

## Test Plan

* Create a failing build and receive an email

## Documentation Plan

* Update the on-call handbook to expect these emails and what to do with them

## Release Plan

* check in github workflows to colcon-bundle

## Acceptance Criteria

* [ ] When colcon-bundle or colcon-ros-bundle scheduled end to end tests fail, the on-call receives an email
* [ ] The oncall has a handbook section for what to do about that email

Once all above items are checked, this story can be moved to done.

## Resources

thomas-moulard, 2020-01-24:
Investigate if we can re-use the component in aws-robotics.

piraka9011, 2020-01-31:
I've created an action that does this but in the meantime we can just use a repo specific GH action.
Spoke with Emerson offline to see how we can open source the action in aws-actions or similarly.

piraka9011, 2020-02-11:
Added the cloudwatch action to colcon-bundle in [this PR](https://github.com/colcon/colcon-bundle/pull/133)

piraka9011, 2020-02-14:
The action is hosted in [ros-tooling/action-cloudwatch-metrics](https://github.com/ros-tooling/action-cloudwatch-metrics).
The action is in colcon/colcon-bundle#137 and colcon/colcon-ros-bundle#33.

closed 2020-02-28 by thomas-moulard