ros-tooling/aws-roadmap #174: View MessageMetrics in RViz

dabonnie, 2020-01-31:
## Description

Implement a plugin to view the [system_metrics_collector](https://github.com/ros-tooling/system_metrics_collector) [MetricsMessage](https://github.com/ros-tooling/system_metrics_collector/blob/master/metrics_statistics_msgs/msg/MetricsMessage.msg) in rviz.

Currently the only visualization exists using the [CloudWatch Metrics ROS2 agent](https://github.com/aws-robotics/cloudwatchmetrics-ros2).

## Test Plan

* Manual testing
  * ensure multiple MetricsMessage produces can be independently viewed

## Documentation Plan

* README.md update
  * provide screenshots

## Release Plan

* check in PR

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

emersonknapp, 2020-01-31:
probably more appropriate for rqt since it's not a 3d world thing?

dabonnie, 2020-01-31:
> probably more appropriate for rqt since it's not a 3d world thing?

Yes, my mistake. Editing accordingly.

thomas-moulard, 2020-04-02:
We have a PR in limbo which could help: https://github.com/ros-visualization/rqt_plot/pull/44

emersonknapp, 2020-07-24:
Dropping for now until we get requests for this

closed 2020-07-24 by emersonknapp