ros-tooling/aws-roadmap #188: Port ROSMovingAverageStatistics to system_metrics_collector

dabonnie, 2020-02-14:
## Description

Port the base class ```ROSMovingAverageStatistics``` to use the messages and common utilities found in the [system_metrics_collector](https://github.com/ros-tooling/system_metrics_collector). This base class provides ROS2 functionality for all of the measurement classes.

Long term we envision this class to be part of a separate package. Specifically, system_metrics_collector and the ensuing topic_statistics packages should both depend on a more generic utilities package.

## Test Plan

* Port existing unit tests for the class
* End to end integration test calling CLI entrypoint
  * Use the system_metrics_collector e2e as a base 

## Documentation Plan

* README.md update
* Document classes

## Release Plan

* check in PR
 * put code in another folder in the system_metrics_collector project

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

thomas-moulard, 2020-02-17:
@dabonnie check whether we can have e2e tests for this one or not.

dabonnie, 2020-02-18:
Updated issue given findings with @prajakta-gokhale.

dabonnie, 2020-02-24:
Note: during implementation of https://github.com/ros-tooling/system_metrics_collector/pull/94/files, this class evolved into a pure interface. Testing is accomplished through the implemented classes.

closed 2020-02-28 by thomas-moulard