ros-tooling/aws-roadmap #193: Update tutorials to highlight that one should not call services from callbacks

thomas-moulard, 2020-02-14:
**REMOVE-ME: Fill out this and all below sections, and remove the PROMPT questions for readability.**

## Description

PROMPT (REMOVE-ME): A concise, concrete, finishable, description of the feature to be built.

## Test Plan

PROMPT (REMOVE-ME): What tests should be created to support this feature?

* Manual testing
* Unit test for new classes
* End to end integration test calling CLI entrypoint
* ...

## Documentation Plan

PROMPT (REMOVE-ME): What documentation updates need to be made to support this feature? e.g.
* README.md update
* wiki update
* design doc PR
* ...

## Release Plan

PROMPT (REMOVE-ME): What release work needs to be done to finish this feature? e.g.

* check in PR
* bloom release

## Acceptance Criteria

PROMPT (REMOVE-ME): Clearly defined checklist of things that need to be done to call this task complete. Can include items from all above sections.

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

PROMPT (REMOVE-ME): Add here links to any relevant document e.g. design (and make sure everyone can access the docs linked here!):

* [FIXME](http://...)

thomas-moulard, 2020-02-14:
@mm318 Please groom this ticket.

mm318, 2020-02-20:
Draft pull request opened at https://github.com/aws-ros-dev/ros2_documentation/pull/2.

mm318, 2020-02-20:
Real pull request opened at https://github.com/ros2/ros2_documentation/pull/538 :smiley:

thomas-moulard, 2020-02-20:
That was a quick LGTM, thanks @mm318  ;)

mm318, 2020-02-25:
Additionally opened a pull request for the `rclpy` API docs at https://github.com/ros2/rclpy/pull/516 to add a stronger warning about not calling services from callbacks.

mm318, 2020-02-27:
OSRF is willing to take over https://github.com/ros2/ros2_documentation/pull/538, and https://github.com/ros2/rclpy/pull/516 is merged.

This ticket may be considered done.

closed 2020-02-28 by thomas-moulard