ros-tooling/aws-roadmap #195: Add a warning to `ros2 launch` when launching multiple nodes with the same name

emersonknapp, 2020-02-19:
## Description

This is a tracking duplicate of https://github.com/ros2/launch_ros/issues/124

Node name uniqueness is not enforced in ROS2. It seems to be uncertain whether-or-when it will be so. In the meantime, there is a lot of user confusion that can happen around having multiple nodes with the same name, since much of ROS2 assumes they are uniquely named.

One user-helping mitigation we can introduce is: Within a single invocation of `ros2 launch`, log a warning message for duplicate node names before actually starting them. This will highlight one source of duplicate node names.

Limitations:
* this does nothing to help in separately-launched launchfiles (e.g. other robots on the network or separate shells)
* this probably can't introspect the default names if users do not specify a node name on the Node action

#### Implementation considerations
I'm not yet familiar enough with this code to understand how to implement. I assume at some point launch has a fully substituted flattened list of things to start, at which point we'd have all the node names expanded out and can detect duplicates.

## Test Plan

* launch_test based test, checking stdout for the warning

## Documentation Plan

* The warning is self-documenting, no additional documentation needed

## Release Plan

* merge PR to upstream

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

External issue: https://github.com/ros2/launch_ros/issues/124

thomas-moulard, 2020-02-20:
Additionally, we should warn users that it is preferable to explicitly name nodes to avoid naming conflicts.

mm318, 2020-02-25:
Opened pull request that implements the warning at https://github.com/ros2/launch_ros/pull/127.

closed 2020-03-13 by dabonnie