ros-tooling/aws-roadmap #202: rosbag2 cannot play back multiple compressed files

dabonnie, 2020-02-24:
## Description

Fix the bug, detailed here https://github.com/ros2/rosbag2/issues/296, where multiple compressed files cannot be played back

## Expected Behavior
rosbag2 can play back multiple compressed files

## Actual Behavior
rosbag2 throws an exception and halts playback 

## To Reproduce
PROMPT: Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Screenshots
PROMPT: If applicable, add screenshots to help explain your problem.

## System (please complete the following information)
 - OS: [e.g. Ubuntu Bionic]
 - ROS2 Distro: [if not described via VCS .repos file, e.g. Dashing]

PROMPT: For ROS2 bugs - please attach a VCS.repos or Dockerfile if possible to help us reproduce the environment.

## Additional context
PROMPT: Add any other context about the problem here.

dabonnie, 2020-02-24:
Potential fix in progress, needs tests added: https://github.com/ros2/rosbag2/issues/296

thomas-moulard, 2020-02-28:
Closing as duplicate of #192.

closed 2020-02-28 by thomas-moulard