ros-tooling/aws-roadmap #210: [1d] Investigate if CycloneDDS supports QoS mismatch callbacks

mm318, 2020-03-04:
## Description

This is required to complete the https://github.com/ros-tooling/aws-roadmap/issues/115 epic.

As mentioned in https://github.com/ros2/ros2/issues/822, FastRTPS (and consequently `rmw_fastrtps_cpp`) currently does not support this (being tracked at https://github.com/eProsima/Fast-RTPS/issues/852). RTI Connext does support these callbacks, but work probably still needs to be done in `rmw_connext_shared_cpp` and RTI Connext is a non-free offering. OpenSplice also does support these callbacks, but it is now irrelevant.

We need to look at our last option, CycloneDDS and `rmw_cyclonedds_cpp`.

## Spike Timeboxing

1 day

## Spike Outcome

- A short document on:
  - Whether or not using CycloneDDS/`rmw_cyclonedds_cpp` is feasible for completing the epic
  - What the behavior should be when a user uses a `rmw` implementation that does not support this feature

## Goals

- Decide whether or not to prioritize the https://github.com/ros-tooling/aws-roadmap/issues/115 epic, as its completion depends on the capabilities of the available `rmw` implementations

mm318, 2020-03-04:
Pushing out the work on https://github.com/ros-tooling/aws-roadmap/issues/125, and pulling in this spike in its place, as the feasibility of the https://github.com/ros-tooling/aws-roadmap/issues/115 epic may affect the outcome of https://github.com/ros-tooling/aws-roadmap/issues/208.

emersonknapp, 2020-03-05:
#115 won't affect #208 - we only need the query API, which has already been merged

mm318, 2020-03-12:
Decently detailed user stories https://github.com/ros-tooling/aws-roadmap/issues/219 and https://github.com/ros-tooling/aws-roadmap/issues/220 have been created as a result of this spike. A formal document on implementation details to follow, if necessary.

closed 2020-03-30 by emersonknapp