ros-tooling/aws-roadmap #225: Define Topic Statisics class in rclcpp

dabonnie, 2020-03-18:
## Description

Define a SubscriberTopicStatistics class, in a separate folder in rclcpp, that depends on [``libcollector```](https://github.com/ros-tooling/aws-roadmap/issues/224). This class will live in the [subscription_base](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/subscription_base.hpp), where the message (and clock time) can be passed to the topic statistic collectors via [handle_message](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/subscription.hpp#L221).

The [received message age](https://github.com/ros-tooling/system_metrics_collector/blob/master/system_metrics_collector/src/topic_statistics_collector/received_message_age.hpp) and [received message period](https://github.com/ros-tooling/system_metrics_collector/blob/master/system_metrics_collector/src/topic_statistics_collector/received_message_period.hpp) collectors are part of this class. 

## Test Plan

* Manual testing
* Unit test for new classes
* should be demo-able, e.g., print debug statements of collected statistics to the console

## Documentation Plan

* class documentation

## Release Plan

* check in PR to aws-ros-dev/rclcpp fork

## Acceptance Criteria

PROMPT (REMOVE-ME): Clearly defined checklist of things that need to be done to call this task complete. Can include items from all above sections.

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* https://github.com/ros-tooling/aws-roadmap/issues/224

dabonnie, 2020-04-16:
Opened meta-ticket to track work: https://github.com/ros2/ros2/issues/901 (thanks for the suggestion @emersonknapp )

emersonknapp, 2020-04-24:
Reviewed at Sprint #17 review - demo node will be separate story under same epic

closed 2020-04-24 by emersonknapp