ros-tooling/aws-roadmap #232: Bloom libstatistics_collector package

dabonnie, 2020-04-02:
## Description

Topic Statistics (in rclcpp) and the system_metrics_collector packages require the libstatistics_collector package. libstatistics_collector should be bloom released.

## Test Plan

* Manual testing
  * test the artifacts on the shadow repo
* End to end integration test calling CLI entrypoint
  * ensure cloudwatch metrics works with the new message

## Documentation Plan

* README.md update
  * ensure readme is properly documented on how to use the package

## Release Plan

* bloom release

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

prajakta-gokhale, 2020-04-28:
Related to https://github.com/ros-tooling/libstatistics_collector/issues/13

dabonnie, 2020-04-29:
Related (and merged): https://github.com/ros/rosdistro/pull/24647

thomas-moulard, 2020-05-02:
Re-opening, let's wait to verify that it's actually working, currently the builds are broken because deps are missing:

http://build.ros2.org/view/Fsrc_uF/job/Fbin_uF64__libstatistics_collector__ubuntu_focal_amd64__binary/

http://build.ros2.org/view/Fsrc_uF/job/Fbin_ubhf_uFhf__libstatistics_collector__ubuntu_focal_armhf__binary/

http://build.ros2.org/view/Fsrc_uF/job/Fbin_ubv8_uFv8__libstatistics_collector__ubuntu_focal_arm64__binary/

dabonnie, 2020-05-05:
Re-bloomed using `bloom-release 0.9.7`, PR here: https://github.com/ros/rosdistro/pull/24727

prajakta-gokhale, 2020-05-08:
Need to verify if the build is stable.

dabonnie, 2020-05-27:
Opened https://github.com/ros/rosdistro/pull/25122 per https://github.com/ros-tooling/libstatistics_collector/issues/13#issuecomment-634384496

emersonknapp, 2020-06-17:
This seems done @dabonnie @prajakta-gokhale @thomas-moulard is this still in progress?

dabonnie, 2020-06-17:
Given the foxy release and the testing from source I thought this was closed (as was agreed in the retro). 

Closing now.

closed 2020-06-17 by dabonnie