ros-tooling/aws-roadmap #233: Add a default incompatible QoS callback for rclcpp users

mm318, 2020-04-02:
## Description

If a user creates a publisher/subscription without specifying a custom incompatible QoS callback, a default one will be used under the hood to deliver notifications (warning messages) of participants found on the topic that have incompatible QoS policies.

## Test Plan

* Manual testing
* Unit test that listens on /rosout

## Documentation Plan

* Warning message is self documenting

## Release Plan

* Merge PR into upstream

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* The code changes will be somewhere [here](https://github.com/ros2/rclcpp/blob/01a6befdde7f78b6f7788e6180cc5ed38390e955/rclcpp/include/rclcpp/publisher.hpp#L76) for the publisher
* The code changes will be somewhere [here](https://github.com/ros2/rclcpp/blob/01a6befdde7f78b6f7788e6180cc5ed38390e955/rclcpp/include/rclcpp/subscription.hpp#L111) for the subscription

mm318, 2020-04-02:
This is also tracked here: https://github.com/ros2/rclcpp/issues/1048

mm318, 2020-04-08:
Relevant pull request opened at https://github.com/ros2/rclcpp/pull/1051.

mm318, 2020-04-08:
Now there is some push back against having this feature at all.

mm318, 2020-04-14:
> Relevant pull request opened at [ros2/rclcpp#1051](https://github.com/ros2/rclcpp/pull/1051).

The relevant pull request has been merged.

emersonknapp, 2020-04-24:
Reviewed at Sprint #17 review

closed 2020-04-24 by emersonknapp