ros-tooling/aws-roadmap #243: [M] Update Quality of Service Wiki with new features since Dashing

emersonknapp, 2020-04-09:
## Description

https://index.ros.org/doc/ros2/Concepts/About-Quality-of-Service-Settings/

The existing wiki page does not completely cover the Quality of Service concepts in ROS 2. Missing concepts:
* Offer and Request semantics (multiple connection)
* Compatibility strictness ordering
* Lifespan
* Liveliness (+event callback)
* Deadline (+event callback)
* Incompatibility callbacks

## Spike Outcome

Updated wiki page PR merged

mm318, 2020-05-07:
The relevant pull request that updates the Wiki page is opened at https://github.com/ros2/ros2_documentation/pull/661.

mm318, 2020-05-11:
The relevant pull request https://github.com/ros2/ros2_documentation/pull/661 has been merged.

mm318, 2020-06-05:
The updated documentation has been published: https://index.ros.org/doc/ros2/Concepts/About-Quality-of-Service-Settings/

closed 2020-06-05 by mm318