ros-tooling/aws-roadmap #245: [S] Add Topic Statistics in the ROS 2 concepts wiki

prajakta-gokhale, 2020-04-09:
## Description

Topic Statistics will be a new feature introduced in ROS2-F. We want users to know about this feature and have a basic understanding of what it does.

Add a Topic Statistics section in [ROS 2 Concepts](https://index.ros.org/doc/ros2/Concepts/) describing what the feature is and what to expect from it.

## Spike Timeboxing

Small

## Spike Outcome

A new section `Topic Statistics` in https://index.ros.org/doc/ros2/Concepts/

## Goals

Explain a new feature introduced in ROS2-F.

emersonknapp, 2020-06-12:
@dabonnie note that the ros2_documentation PR should be linked to this issue https://github.com/ros2/ros2_documentation/issues/722

dabonnie, 2020-06-19:
Can see the formatted file (merged into master) [here](https://github.com/ros2/ros2_documentation/blob/master/source/Concepts/About-Topic-Statistics.rst), not sure how often https://index.ros.org/doc/ros2/ is built.

emersonknapp, 2020-07-17:
@dabonnie to change "empty data" to "nan" in https://index.ros.org/doc/ros2/Concepts/About-Topic-Statistics/

emersonknapp, 2020-07-24:
Done

closed 2020-07-24 by emersonknapp