ros-tooling/aws-roadmap #246: [M] Add Topic Statistics Tutorial in ROS 2 wiki

prajakta-gokhale, 2020-04-09:
## Description

Topic Statistics will be a new feature introduced in ROS2-F. We want to demonstrate how to use it.

Add a new tutorial in [ROS2 Tutorials](https://index.ros.org/doc/ros2/Tutorials/) demonstrating how to use topic statistics. What the tutorial may include:
1.  How to enable/disable topic statistics?
2. How to discover parameters that can be tuned?
3. How to change these parameters?
...

## Spike Timeboxing

Medium

## Spike Outcome

Tutorial for topic statistics

## Goals

Enable users to use topic statistics

dabonnie, 2020-06-22:
Added outline [here](https://github.com/aws-ros-dev/ros2_documentation/tree/dabonnie/topic-statistics-tutorial). Note how this is similar to the ROS2 Pub/Sub tutorial. The following needs to be completed:

 - Finish subscriber node with subscription options to enable topic statistics: https://github.com/aws-ros-dev/examples/tree/dabonnie/topic-statistics-examples
 - Use the example publisher (chatter / string) to collect statistics for data without a header timestamp
 - Add a publisher / subscriber that uses a message with a header timestamp to collect statistics
- Update tutorial: https://github.com/aws-ros-dev/ros2_documentation/tree/dabonnie/topic-statistics-tutorial

emersonknapp, 2020-07-24:
Done

closed 2020-07-24 by emersonknapp