ros-tooling/aws-roadmap #247: Move our packages' CI to Ubuntu Focal

emersonknapp, 2020-04-10:
## Description

Foxy uses Focal - we need our CI for packages we're releasing to use it
* rosbag2
* rosbag2_bag_v2
* libcollector

## Test Plan

* CI passes when using Focal

## Documentation Plan

N/A

## Release Plan

* Merge PR changing Action CI settings (when green)

## Acceptance Criteria

* [ ] Code has been implemented, reviewed and merged.
* [ ] Test plan has been completed
* [ ] Release plan has been completed

Once all above items are checked, this story can be moved to done.

## Resources

* [FIXME](http://...)

thomas-moulard, 2020-05-02:
rosbag2 PR in review: https://github.com/ros2/rosbag2/pull/401

thomas-moulard, 2020-05-02:
@prajakta-gokhale I think it's unreasonable to bundle rosbag2_bag_v2 in this user story, as this CI never worked. Would it be OK to have a follow-up task an extract the subpart of this user story which is more complex. Updating rosbag2_bag_v2 should be blocked on making the CI work first.

prajakta-gokhale, 2020-05-06:
That's fine by me.

prajakta-gokhale, 2020-05-07:
@thomas-moulard seems like the last PR from rosbag2 is merged. Is there anything else left for this task?

closed 2020-05-22 by thomas-moulard